JvanKatwijk/qt-dab #314: Spyserver integration

sdrnerd, 2024-01-03:
Can the use of a Spyverter be integrated? It can be used by all Airspy's and RTL devices.
I like to mount an Airspy directly on my antenna with a Raspberry Pi running the Spyserver to avoid coax losses, hence getting better results.

JvanKatwijk, 2024-01-04:
I do not know anything about spyserver,  so if you can tell me what the
protocol is that it uses I can have a look
(Qt-DAB supports rtl_tcp )

andimik, 2024-01-04:
Dear @sdrnerd 

You could use a `fifo` where you (virtually) store your spyserver rawfile. This you can load in Qt-DAB.

I'm doing similar thinks for many years:

```
mkfifo foo.raw
```

In `odr-dabmod` I am using this fifo file as output and can then open a virtual raw file in Qt-DAB.

In your case you need to fetch the file from your raspberry, as far as I have understood.

JvanKatwijk, 2024-01-05:
Well, I found some client software and will have a look at it. Be aware
however that a wireless connection betwen a separate servicer and a client
program requires quite some bandwidth: assuming an rtlsdr device on the
serverside over 2 * 2048000 bytes/second, assuming an Airspy (or something
like that)  the required bandwidth will double.

If you have a wired connection, that will not be a problem of course

sdrnerd, 2024-01-05:
Jan, Spyserver is a development of Youseff Touil, the maker of the Airspy
SDR's and also author of SDR#. It also contains support for RTL-devices and
can recognise the connected SDR's automatically. It is a kind of
development derived from RTL-TCP. With this the Airspy models can be used
and at the same time have RTL sticks connected.

Look on the Airspy pages for more details:
Airspy SDR - High Quality Software-Defined Radio, Redefined
<https://airspy.com/>

You can find the spy-server on the download page and also info on the Quick
Start page.

Regards,
Ron Liekens
ON2RON

JvanKatwijk, 2024-01-06:
The issue is what the communication protocol is for the spyserver. I found
on the internet a client that might give some of the answers
As far as I can see, it does only integer decimation, so if you use e.g. an
Airspy, the required bandwidth is 4 * 2500000 bytes/second (not counting
all kinds of control data that goes up and down)

JvanKatwijk, 2024-01-12:
If you use Linux then I do have a - more or less - working client - based
on the client code I found on github.
If, however, your stuck to Windows then I am afraid that while the Linux
solution compiled fine for Windows,
the Windows networking causes a terrible package loss

JvanKatwijk, 2024-01-16:
With some help from Youssef (Airspy) I created a (very first) version for
Windows. In the downlod section of the repository, one will find an
AppImage for use under Linux and a  windows installer, the 32-6.40 version,
bith have spyserver as an ebtry in the device table
No garantees however

best
jan

closed 2024-01-20 by JvanKatwijk