JvanKatwijk/qt-dab #315: raspberry error on cmake

davidea72, 2024-01-10:
hi guys, i'm trying to compile qt-dab6 on a raspberry with raspbian 12, but i'm stuck on a Cmake error

CMake Error at CMakeLists.txt:114 (message):
  please install qwt library

i htink it's
libqwt-qt5-dev - Qt widgets library for technical applications (development, qt5), but it's at the newest version (6.1.4-2)

maybe the check is for an old version? unfortunately i'm not so smart to correct the cmake file, can someone help me?

mfraser, 2024-02-15:
Don't use cmake, use qmake instead.
Edit qt-dab-6.pro to uncomment
`LIBS		+= -lqwt`
You can also comment out any devices below that you don't need.
`mkdir build && cd build`
`qmake ..`
`make`

andimik, 2024-02-15:
@davidea72

Maybe you try `ccmake`, which is like a "GUI" for `cmake`. It will show all options.

JvanKatwijk, 2024-02-15:
There is a known difference of the location for qwt stuff between debain
based systems (e.g. ubuntu) and Fedora based systems (that I use).
I use Ubuntu for generating the Windows versions, and I try to make the
"pro" file such that I can use the same file for both
Fedora and Ubuntu.

I always advise to use qmake/make after doing some editing to the ".pro"
file. Development by me is always dome with the ".pro" file at hand,
and usually, after some time I (try to) update the CMakeLists.txt file.
The ".pro" file contains (far) more possibilities for selecting an option

zygmund2000, 2024-03-25:
I have same problem on Arch Linux and how to fix that?