JvanKatwijk/qt-dab #317: MOT Plaatje { Dia }

dabdxjan, 2024-01-25:
Goedenmiddag 
Paar dagen terug qt-dab 64 { 64 bits } gedownload en geinstalleerd , 
alles werkt prima , maar ik had 1 probleem dat is met de MOT Plaatjes ,
Ik krijg geen MOT Plaatjes {Dia's } meer te zien op verschillende Radio Station's ? 
Met qt-dab32 6.25 ontvang ik wel alle MOT Plaatjes van alle Radio Station's .
73s
Jan 
sdrplay-1 { eerste versie } 
Antenne 7 Elements Fuba { verticaal }

JvanKatwijk, 2024-01-25:
Helaas, dat ligt aan de plugins die voor de gegeven 64 bit versie te vinden
zijn. Voor sommige plaatjes in JPG format is geen passende decoder plugin
in de
verschillende Qt libraries te vinden.

dabdxjan, 2024-01-25:
Dat is Jammer Jan , voor de rest werkt qt-dab 64 64 bits uitstekend ! 
Helaas kan ik de laatste qt-dab-6 beta niet installeren want mijn virusscanner Eset Smart Security Premium laat het niet toe , 
ik kreeg en melding van een Trojaans paard in qt-dab32-6 beta ?? 
[image]
Jan

JvanKatwijk, 2024-01-25:
Bij de vorige versie(s) war er ook al gedoe van virusscanners, dat is de
reden dat bij de compilatie de "optimization" vlaggen niet op maximaal staan
In een van de Qt libraries zal - na maximale optimalisatie van de code -
een patroon dat als virus gezien werd.

Ik zal kijken of ik de installingen zo kan zetten dat er geen
waarschuwingen meer komen, maar ik werk onder Linux en cross compileer voor
Windows, dus dat kan even duren

dabdxjan, 2024-01-25:
Bedankt Jan 
met downloaden van qt-dab 6.25 had ik ook een berichtje gehad van Eset virusscanner , er was op de achtergrond een Suspicious verwijderd ?? .
maar gelukkig werd qt-dab 32 6.25 niet verwijderd , 
ik wacht rustig af Jan 
Alvast bedankt !! 
73
Jan

JvanKatwijk, 2024-01-28:
Ik heb de installerfile noh maar eens een keer door "virustotal"gehaald en
daar komt alleen ündetected"uit, dus ik weet even niet hoe ik hier verder
mee moet
vriendelijke groet
jan

dabdxjan, 2024-01-28:
Goedenmiddag jan 
Het is jammer , ik weet het ook niet helaas , gelukkig kan ik qt-dab 6.25 gebruiken voor Dab dxen , 
als er weer updates komen voor qt-dab dan probeer ik weer te downloaden en mischien lukt het dan wel .
Alvast Bedankt Jan voor het proberen te oplossen van deze probleem !  
Groeten
73s
Jan

JvanKatwijk, 2024-01-28:
De wijzigingen in de laatste versies hebben primair te maken met (a) de
scan control, en (b) de presets of favorites
Bij het scannen - dat op zich niet gewijzigd is - is er in de laatste
versies een apart control widget en de scan button regellt
de sichtbaarheid van dat widget,
Bij de presets is er in eerste instantie gekozen voor een aparte lijst ipv
verstopt achter een combobox, maar in de laatste versie
zijn ze zichtbaar gemaakt in de main widget.
Beide niet direct wijzigingen die met DX-en te maken hebben dus

Vriendelijke groet
jan

dabdxjan, 2024-01-28:
Hoi Jan 
ik ben even bij Eset virusscanner in logfile's geweest en kwam deze melding tegen toen ik qt-dab 6.25 ging installeren , 
gelukkig werd de qt-dab 6.25 niet verwijderd na deze
[image]
 actie van eset , ik heb een screen foto van gemaakt , 
Hopelijk met nieuwe updates dat het wel gaat lukken jan . 
Groeten
Jan

dabdxjan, 2024-01-28:
Bedankt voor de bericht Jan 
Groeten
Jan

dabdxjan, 2024-02-02:
Goedenmiddag Jan 
ik wou even doorgeven dat de laatste versie update qt-dab64 { 64 bits } prima werkt ! , 
geen problemen met de virusscanner ! , downloaden en installeren ging uitstekend ,
hopelijk dat in de toekomst dat MOT Plaatjes { dia's } voor alle radio's station's  gaan werken in qt-dab64 
ik vind het wel jammer dat tii code is verdwenen op de voorgrond , 
ik maak veel screen foto's van dx ontvangsten en dan is het mooi dat ook de tii code erop staat , 
nu heb je onderin in het voorgrond een station's locatie maar geen tii code dat is wel jammer , 
voor de rest is qt-dab een prachtige software !! 
Groeten 
73s
Jan

JvanKatwijk, 2024-02-02:
Het toeval wil dat ik net zat te experimenteren met het plaatsen van de
(mainId, subId) op de main widget. Probleem is natuurlijk dat
het een gegeven is over de input, dus het is niet logisch om het bij de
gekozen service te zetten. Bij de ensemblenaam is er niet genoeg ruimte,
De oplossing die mij wel aardig leek is de (mainId, subId) combinatie op de
regel te zetten waar ook de zendernaam komt te staan, maar
dan zonder begeleidende tekst, dus alleen (04 05) oid

Dat heb ik in de 64 bit versie (en de Linux AppImage) gezet en die versie
ge-upload

Nog een opmerking over virusscanners: alle versies zijn met dezelfde
toolset en de zelfde set sources ge"bakken", de 32 bit versies delen ook
nog alle libraries (behoudens de rtltcp library, die is voor de V3 en 6.40
versie verschillend. Een visusscanner test op bitpatronen en in een wat
groter
programma als Qt-DAB komen diverse bitpatronen voor.
De Windows 6.5Beta versie waarbij een stevige optimimalisatie werd
toegepast gaf bij verschillende virusscanners een "besmet" signaal,
toen de optimilisatie volledig uitgeschakeld werd viel die ontdeking in
grote bij o.a. Defender weg.
Als 6.40 geen meldingen geeft is dat een (m.i.) overtuigend bewijs dat de
scanner die "iets" vindt in 6.5Beta een false positive geeft,
de twee versies zijn qua source meer dan 95% gelijk en gebruiken  precies
dezelfde libraries

Ik kan dus verder niets verhelpen aan de geconstateerde besmetting want het
is een false positive

Overifens is het prettig dat er geen melding op de 64 bit versie komt

Vriendelijke groet
jan

dabdxjan, 2024-02-02:
Hoi Jan
ja klopt het is zeker een 100% false positieve melding van Eset ! , 
met downloaden van qt-dab 6.25 kreeg ik een melding van Eset dat er een suspicious virus werd verwijderd , 
maar qt-dab32 6.25 werd deze keer niet verwijderd , dat vond ik al heel apart , toen wist ik al dat het false meldingen waren , 
gelukkig werd qt-dab32 625 niet verwijderd , 
mooie oplossing Jan met zendernaam en daarachter alleen de nummer van de tii code .
is deze update qt-dab64 { 64 bit } met de nummer van de tii code nu te downloaden ? .
Alvast Bedankt 
73s
Jan

dabdxjan, 2024-02-02:
Goedenavond Jan 
Bedankt voor de tii code bij de zendernaam Top , 
Groeten 
73s 
Jan 
[image]

JvanKatwijk, 2024-02-03:
Mooi

Overigens zie ik dat als je geen gekleurde "skin" hebruikt, sommige van die
icoontjes wel behoorlijk onzichtbaar worden.
Ik gebruik zelf altijd de fefault skin.
Verder is het opvallend dat bij de 64 bit versie de randen van de tabel
zichtbaar zijn, bij zowel de 32 bits versies als de AppImage
zie je die niet.

Aarzel verder niet om vragen oid te stellen, ik vind het juist oo leuk om
aan/met de programamtuur te knutselen

Vriendelijke groet
jan

dabdxjan, 2024-02-03:
Goedenmiddag jan 
Bedankt voor de Tip van gekleurde skin! 
ik zag Fefault Skin niet in mapje skin staan , wel anderen ,
Darkeum skin heb ik nu gekozen en mooi hoor , nu is alles mooi te zien ! 
qt-dab64 64 bit is prachtige software TOP ! 
Als MOT Plaatjes is opgelost is het helemaal Super 
Bedankt Jan 
Groeten
73s
Jan 
[image]

JvanKatwijk, 2024-02-03:
Dan weet je waarschijnlijk ook niet dat je de kleur van de verschillende
knoppen (de PushButtons) kan instellen door met de rechtermuisknop aan te
klikken,
hetzelfde voor de elementen in de display.
Zelf wil ik altijd in verschillende displays verschillende kleuren hebben
Van de list van services in de hoofdwidget kan je het font kiezen, de
fontgrootte instellen en de fontkleur kizen, zelf gebrui ik daar cantarell
12pt met als kleur cyan

Dank voor je compliment, het is een leuk project waar ik nog steeds veel
van leer

vriendelijke groet
jan

dabdxjan, 2024-02-07:
Goedenavond Jan 
Bedankt voor de nieuwe Update !! 
Alles werkt weer perfect Downloaden en Installeren qt-dab 64 { 64bits } ,
Kleurenpallet werk Top Jan ! 
ik heb een screen foto bij gedaan 
Groeten 
73s
Jan 
[image]

dabdxjan, 2024-02-07:
Hoi Jan 
inmiddels de kleuren weer veranderd 
Groeten
73s
jan 
[image]

JvanKatwijk, 2024-02-07:
Een beetje kleur is altijd wel prettig, leuk om te zien.
Vriendelijke groet
jan

dabdxjan, 2024-02-28:
Goedenmorgen Jan 
Bedank weer voor de update van qt-dab 64 { 64bits } .
qt-dab 64 werkt uitstekend ! , 
Overdag luister ik meestal met qt-dab 64 op de achtergrond als ik met de computer bezig bent en met de Icom-r8600 ,
hopelijk in de toekomst dat met de dia's { Mot Plaatjes } is opgelost met qt-dab 64 { 64bits} , 
ik heb gisteravond een screen foto gemaakt .
Bedankt !! 
Groeten
73s
Jan 
[image]

dabdxjan, 2024-04-05:
Goedenavond Jan 
Herman had melding gedaan dat nieuwe qt-dab32-6x  is uitgebracht .
ik heb qt-dab 32-6x gedownload en geinstalleerd , 
en zeer verrassend was het wel , dat ik geen problemen had met de Eset viruscanner ,
mischien komt het wel door dat qt-dab andere bestandsnaam heeft qt-dab32-6x en geen exe meer erachter .
ik zeg maar wat Jan hoor , ik weet het zelf ook niet , alles is gelukkig goed gegaan en dat is belangrijker .
Dx Display Widget is ook leuk , ik heb getest met lokale station's en leuk om te zien welke zender je eigenlijk ontvangt ,
ik zal met dxen dx display gebruiken en ook weer even uitschakelen voor een screen foto maken van ontvangen zender , 
voor een screen foto is het prachtig dat op de hoofd Widget de zender en tii opstaan , dan heb alles op 1 foto staan ,
ik heb een screenfoto van QT-DAB 32-6X erbij gedaan .
Bedankt Jan 
73s
Jan
Kreileroord

dabdxjan, 2024-04-05:
[image]

closed 2024-04-05 by dabdxjan