JvanKatwijk/qt-dab #318: Version 6.5 Beta?

dwarning, 2024-02-25:
I made a clone of the github:
git clone https://github.com/JvanKatwijk/qt-dab.git
So I got a subdir qt-dab-s5 and was able to build a version 5.5.
But how I can build the higher versions like 6.4 or 6.5 beta?
There are no qt-dab-s6 or similar in my tree.

JvanKatwijk, 2024-02-25:
try qt-dab-RC

andimik, 2024-02-25:
Is this new, Jan?

dwarning, 2024-02-26:
Thank you - now it works better.
cmake build struggling over missing qwt lib, which I have installed. 
The I went to qmake which build fine. But I have no glue how to install: "sudo make install" has no target in the makefile. Copying binary, desktop and icons per hand is OK.

Seems the doc Readme and .pdf needs few updates.

JvanKatwijk, 2024-02-27:
No it was in the process of creating a 6.5 version, the "development"
version is put in the "RC" (Release Candidate) directory

closed 2024-10-17 by JvanKatwijk