JvanKatwijk/qt-dab #319: SDRplay API 3.14 ?

Dirk275, 2024-02-27:
Short question:
Is Qt-DAB 6.5 Beta compatible with the new SDRplay API 3.14?
I ask, because my RSP1 doesn't work with Qt-DAB in a manner, that it is recognized, but all changes in the device window "SDRplay control-v3" (like ppm control, agc, lna sate selector ...) seem to have no effect and a reception of a program saved before with another device in a correct scanlist is not possible.

Is the RSP1 known to work with Qt-DAB and the SDRplay API 3.14?
(in SDRuno my device works fine with this API)

JvanKatwijk, 2024-02-27:
Do you mean  an RSP1 or an RSP1a
RSP1 is definitely not supported with the 3.14 API
The RSP1a works fine - both under Linux and windows - with the 3.XX series.
Noet that the Qt-DAB gui has two entries, one for use with the 2.XX
libraries (mainly for use
under Linux or Windows if no 3.XX (XX >= 10) is installed

Dirk275, 2024-02-27:
Thank you Jan!
I think it is an RSP1 compatible China clone.
But it works with the actual SDRuno.
Would it help to install an older SDRplay API 2.xx or e.g. 3.07 to make it work with Qt-DAB?

JvanKatwijk, 2024-02-27:
I assume you are working with Windows. If  it works with SDRuno there
should not be a problem with Qt-DAB.
So, just for my understanding:
a. you are working under Windows, you have installed sdrUno with the 3.14
library. Your clone works fine with sdrUno but not with Qt-DAB?

JvanKatwijk, 2024-02-27:
> Yes, I'm working with Windows 11-23H2 and SDRuno 1.42-1128 with the 3.14
> API.
> The RSP1 clone (with BG7YZF and Huanghuai University PCB logo) works fine
> with SDRuno up to the latest version.
> The SDRplay control-v3 window in Qt-DAB calls it "mirics-SDRplay" and
> displays "3.14" in the upper right corner.
> agc seams to habe no effect. the lna state selector can be set from 0 to
> 99, but the number right of the lna selector field always stays 0.
> Without agc I can also change the if gain reduction from 20 to 59 and left
> it at 20 (or switched agc on).
> Then I tried to play a program from my local scanlist (saved with an
> RTL-SDR Blog V4) and get no signal at all.
> I'm using the same antenna for the RSP1 clone and the RTL-SDR Blog V4.
> Anything I could test?

Thansk that helps to pinpoint what is going on

It seems to be a special kind of RSP 1. Officially, the later 3.XX
libraries do NOT support the RSP1 (And I can assure
you that is correct, I have one), so I never considered support for the
RSP1 in recent rewrites of the driver
I do not support the "RSP1" version in my interface code since RSP1 is not
supported
The "mirics-SDRplay" text is derviced from the device, so apparently  ir
can be opened, but the
dispatching th the section that does the actual work fails. That means that
the Identification code of the
device is probably the RSP 1 identification, but the behavior is perhaps
similar to the 1A

IThe code identification in my code leads to a section handling the
specifics for that RSP, and it is clear to
me that the "default', i.e. empty driver is selected.

What I shall do tomorrow is build a version where the device identification
can be overruled, such that the software
thinks it is a different device

Dirk275, 2024-02-27:
I see!
I don't know if it helps, but as you said "default", i.e. empty driver, I remember, that the main Qt-DAB window had the title "Qt-DAB-6.5.Beta with RTL-SDR-V4" though I actually used the RSP1 clone.
The SDRplay control-v3 window - fitting the RSP1- made me think, I could control the RSP1, but perhaps it was just what you called the "empty driver" and Qt-DAB still assumes the RTL-SDR Blog V4 is the input device!?

JvanKatwijk, 2024-02-27:
You could try the new version. I hacked somewhat and it turns out that I
can use my original RSP1

Dirk275, 2024-02-28:
Now my RSP1 has come to life as well.
The sound is a bit choppy, but this is possibly due to my bad clone.
THANK YOU very much for looking into this!!

Dirk275, 2024-02-28:
Just continued testing:
Choppy sound gone without USB extension cord (how silly!).
Everything fine.

A small thing:
The main window still shows "Qt-DAB-6.5 with RTLSDR-V4" though actually using the RSP1 and I never used the RTL-SDR Blog V4 after updating to your new V6.5. But again: THANKS! everything is fine.

JvanKatwijk, 2024-02-28:
What do the different widgets show
a. the spectrumwidget shows -various aspects if the incoming signal, and
the progress indicator at the bottom tells the quality of the FIC decoding
(FIC channel contains the descrivptive info of the pauload)
b. The technical data widget shows - if you have DAB+)  3 progress
indicator, the first one being the most
important one since that tells how many audo segments are detected and pass
a first test

c. Again, if you have DAB+, the RS error indicator in the technical widget
tells the amount of correction needed (note: on audio carrying segments
that pass the above menstioned first test) on the audio carrying segments

So I would be interested to see a screenshot with the widgets shown

the naming of the versions - there are two 32 bits windows versions differ
in their support for rtlsdr sticks, an apparent
observation was that a using a support library that could handle V4
dabsticks made the software rather deaf when
using iolder sticks. So, one version is equipped with rtlsdr support for V3
and the other with older libraries for V4

Dirk275, 2024-02-28:
Here is a screenshot.
[image]

JvanKatwijk, 2024-02-28:
> Here is a screenshot.
> Screenshot.2024-02-28.091313.png (view on web)

The pocture shows a stromg signal that is processed without errors, in the
technical widget you can see that the number of rs corrections (average
over the last X frames) is zero, so the signal i strong and good. The
percentage of the audio is also 100%, meaning that the rate of the incoming
audiosamples is correct.
One thing you cannot see here is the CPU load, that is shown on the
configuration widget. If that is really high then
you could loose audio fragments in the Windows audio system

I dusted off my old RSP1 and ran for about 15 minutes a test, perfect sound
(but I run Linux)

best
jan

Dirk275, 2024-02-28:
My RSP1 clone also runs perfectly now!

Dirk275, 2024-02-28:
RSP1 added as input device in version 6.5.

closed 2024-02-28 by Dirk275