JvanKatwijk/qt-dab #321: build completed but core dumped

zygmund2000, 2024-03-25:
Hi,

After some time of hard work I compiled latest version qt-dab-6.5 without errors but it can't run:

main: Detected system language "pl_PL"
main: Set language "pl_PL"
main: Error while loading language specifics "pl_PL" use English "en_GB" instead
QWidget: Must construct a QApplication before a QWidget
Stopped (core dumped)

I uncomented in .pro usused hardware like sdrplay lime etc because I want to use only rtl-sdr.
My system is Arch linux.

JvanKatwijk, 2024-03-25:
which OS on RPI

zygmund2000, 2024-03-25:
Not RPI standard x86

JvanKatwijk, 2024-03-25:
Can you send me the "pro" file, then I can exercise. It looks as if some Qt
module cannot be found

zygmund2000, 2024-03-25:
[qt-dab-6.5.txt](https://github.com/JvanKatwijk/qt-dab/files/14747531/qt-dab-6.5.txt)

zygmund2000, 2024-03-25:
You don't need to look at it, because I did it one more time and now works :)

closed 2024-04-05 by JvanKatwijk