JvanKatwijk/qt-dab #324: When rtl_tcp is sucessfully connected, scan stay disabled

cedricpah, 2024-03-31:
When rtl_tcp is sucessfully connected to a host, scan stay disabled and can't manually assign channel.

JvanKatwijk, 2024-04-01:
that is correct
Using rtl_tcp and scanning is not a good combination

JvanKatwijk, 2024-04-01:
if you tell what version/playform you are using I'll create a version for
you that supports scanning with rtl_tcp

cedricpah, 2024-04-01:
I try with lastest 6.5 x64, 6.5x32v3 (because the stick is v3).
It works well with 4.7 (I've try after).

JvanKatwijk, 2024-04-02:
you could give the one where the link points to a try
https://www.dropbox.com/scl/fi/e2d7p7612vyd2ahiom7yt/setup-qt-dab64.exe?rlkey=aplobpib9g95a00gt124ahzkc&dl=0

It is the 64 bit version, and (1) the default setting for the port is 1234
(and setting the port in the .ini file is not overruled anymore)
and (2) the constraint on using the rtl_tcp for scanning is deleted,

good luck

Let me know if it works, then I'll generate 32 bit versions as well

cedricpah, 2024-04-02:
Good : the port is the one of ini file and interface is now enabled.
Thank you. Good job, very well software.

closed 2024-04-02 by cedricpah