JvanKatwijk/qt-dab #327: Compiling for raspberry >= 4 (armv8a)

arkadirenko, 2024-04-13:
Hi,
its again me.  After I have now a reliable DAB-stick, I went a step further and tried to compile qt-dab, as I wished to use this for optimize my antenna position.
By doing so, I run into the same problem discussed here: https://github.com/JvanKatwijk/qt-dab/issues/153

Unfortunately none of the mentioned solutions are working, as I get either complaints about missing neon-headers or definitions, or about missing SSE items.

When searching for the problem, I came to the conclusion, that the problem is, that 64bit Raspberries running armv8a compatible cpus have the NEON command set mandatory and need either special libraries nor special defines.

This [post](https://stackoverflow.com/questions/29851128/gcc-arm64-aarch64-unrecognized-command-line-option-mfpu-neon) showed me a kind of direction, which I could prove working, following the advice found [here](https://forums.raspberrypi.com/viewtopic.php?t=364435#p2186655)

The documentation found in the internet is sometimes unreliable, as -even on webpages from ARM itself- are providing tips without naming the specific ARM version. So the widly found gcc parameter `-mfpu=neon` results with _**gcc** (Debian 12.2.0-14) 12.2.0_ in a `gcc: error: unrecognized command-line option ‘-mfpu=neon’`

So for my understanding, the 64bit Raspis with ARM Cortex A72/A76 are sitting between the chairs, as the understand neither SSE nor the need explicit NEON header or parameters.

So I tried to adapt the qt-dab-5.pro file, uncommented everything for PC and Windows and configured RPI, as this has no Neon settings, but the no-sse sources of viterbi-spiral.
```
RPI     {
        DEFINES         += __MSC_THREAD__
        DEFINES         += __THREADED_BACKEND
        HEADERS         += ../src/support/viterbi-spiral/spiral-no-sse.h
        SOURCES         += ../src/support/viterbi-spiral/spiral-no-sse.c
}
```
But when I compile (qmake; make) I get the following linker error:
```
/usr/bin/ld: /tmp/ccP7Lgj9.ltrans9.ltrans.o: in function `journaline_dataHandler::journaline_dataHandler()':
/home/wmk/DAB-etc/qt-dab/qt-dab-s5/../src/backend/data/journaline-datahandler.cpp:58: undefined reference to `journalineScreen::journalineScreen(std::vector<tableElement, std::allocator<tableElement> >&)'
/usr/bin/ld: /home/wmk/DAB-etc/qt-dab/qt-dab-s5/../src/backend/data/journaline-datahandler.cpp:64: undefined reference to `journalineScreen::~journalineScreen()'
/usr/bin/ld: /tmp/ccP7Lgj9.ltrans9.ltrans.o: in function `journaline_dataHandler::~journaline_dataHandler()':
/home/wmk/DAB-etc/qt-dab/qt-dab-s5/../src/backend/data/journaline-datahandler.cpp:69: undefined reference to `journalineScreen::~journalineScreen()'
collect2: error: ld returned 1 exit status
```

qt-dab.5.pro, uploaded as .txt, as .pro is not supported.
[qt-dab-5.txt](https://github.com/JvanKatwijk/qt-dab/files/14967523/qt-dab-5.txt)

Has anybody an idea?

JvanKatwijk, 2024-04-13:
Actually, I compiled it last week on an RPI3 (took a while but went without
problem)
Which version are you compiling?
As far as I can see, the pro file for the RC version (6.X) contains a clear
reference to the journaline-screen files.
It might be the case that the 6.5 and 5 versions have a problem there.
I was working on an experimental journaline decoder when I got ill, and due
to a problem in the Windows version for 6.X  I uploaded sources for 6.X
but it is well posssible that I forgot to adapt the pro files for versions
6.5 and 5 as well.

I'll have a look

JvanKatwijk, 2024-04-13:
I just checked

The "pro" file for 6.5 was not updated. If you add
     ../includes/backend/data/journaline-datahandler.h \
---->        ../includes/backend/data/journaline-screen.h \
           ../includes/backend/data/journaline/dabdatagroupdecoder.h \

   ../src/backend/data/journaline-datahandler.cpp \
 ---->          ../src/backend/data/journaline-screen.cpp \

then it will compile withoyt problems

As soon as I completed the modification I am applying now to (some of) the
sources, I upload a ciorrected version
(Most likely the CMakeLists.txt files should be updated as well)

arkadirenko, 2024-04-13:
OK, thanks. After adding those two lines in the Version 5 pro and removing the NO_SSE definition also (because of duplicate entries in the RPI definition, compilation went through.

So the recipe for compiling this binary for 64bit Raspis with ARM Cortex A72/A76  is the following:
Use the simple definition RPI, no NEON and neither SSE nor NO_SSE settings. 
I attach my working [qt-dab-5_pro.txt](https://github.com/JvanKatwijk/qt-dab/files/14968615/qt-dab-5_pro.txt) file here.

closed 2024-04-13 by arkadirenko

JvanKatwijk, 2024-04-14:
I always suggest to use the NO_SSE opyion since the choice of RPI options
is limited and strongly depends on the RPI model