JvanKatwijk/qt-dab #328: SDRPlay RSPdx vs RSPduo

Frank-56, 2024-05-30:
Hello,

My RSPdx runs very well with all versions of qt-dab, I usually receive 7 ensembles at my place in NRW. Now when I disconnect the RSPdx and connect the RSPduo, qt-dab does only sync on one ensembles (5C from Dortmund). I can get hardly receive more when I switch agc off and try manual settings. Qirx and SDRUno behave the same on both receivers,

What can be the reason for this? Do you need more information?

Cheers and 73
Frank

JvanKatwijk, 2024-05-30:
Good question. I do not have a duo, so debugging is a little difficult.
Browsing the code I found that I did not set the Duo on single tuner mode,
so I updated that and rebuild the 6X windows version (Assuming
you ar erunning windows).
So you could give that a try.
Gurthermore, for thise channels that you know there is data but with th Duo
you do not get it, do you get a reasonable signal? what does the spectrum
and correlation windows show (compared to the windows for the 5C channel
that you receive_

Frank-56, 2024-05-30:
Dank je wel, Jan, I'll give it a try later, but not setting the tuner mode corectly might well be the reason. And yepp, it's Windows of course.

> do you get a reasonable signal
Yes, the spectrum looks good, but it does not sync.

Cheers
Frank

Frank-56, 2024-05-30:
Couldn't wait ;-) Nope, that was not the reason. I'll do some further checks and get back.

Frank

Frank-56, 2024-05-30:
No problem, Jan. Can do that over the weekend. From both RSPdx and RSPduo or is RSPduo enough? How long should the dump be? 30 seconds?

JvanKatwijk, 2024-05-31:
Looking at the documentation I noticed that the default setting for antenna
in RspDuo is port 2, and there might be an issue
in the code for setting antenna. Did you check with antenna on port 2?

wrt the recording, just 30 seconds of recording of a channel that should
give data

Frank-56, 2024-05-31:
Yes, there is something with the antenna port, but that's not all. A strong station from Dortmund on 11D I can receive after selecting from A to B to C and back to A !? But for a rather weak station (Osnabrück on 8B) this does not work, on RSPdx it's there immediately. I know that Clem Schmidt has extra values for antenna port selection for RSPduo in RSP3_tcp.exe. I also do not find antenna port selection on duo in the API specs 3.15. But I see it's harder to code with # of tuners and master/slave mode etc.

I also found out that on RSPdx the initial "Verstärkungsabnahme" is 37. On RSPduo it's 62 and I cannot select larger values than 10 in the lna setting - on RSPdx I can,

Dumps will follow, must see where to upload those big files.

JvanKatwijk, 2024-05-31:
The number of slectable lna values for the Duo is 10, for the Dx it is much
larger,
The default antenna port selected is "B",  so did tou checj what happens
when the antenna is actually connected to port B?
As said, I do not have a Duo so it is all wild guessing here, but I am
studying the api interface for particularities for the Duo

for transferrig large files you can use wetransfer which is a free service

JvanKatwijk, 2024-05-31:
It might well be that I misinteroreted the antenna settings.
If you set the antenna selector on 'B' and the antenna itself just on the
port for tuner 1, is there then an improvement

Frank-56, 2024-05-31:
Ah, antenna ports on duo are named 1, 2 and HiZ, so I assume 1=A, 2=B, HiZ=C. Anyway, no change which port I use, no change which antenna I use.

I know it's hard to debug with the hardware affected.

Frank-56, 2024-05-31:
I found out, that the "Verstärkungsabnahme" varies on startup and on changing the antenna port. Sometimes I can receive 11D when I change from A to B and back to A. If I do that a seconds time, it won't work.

JvanKatwijk, 2024-05-31:
> Just for my understanding, where do you read the "Verstarkungsabname"?
> The widget for the device handling has an agc selector, an if gain
> reduction section and an lna state selector.
> If agc is selected then the if gain reduction is switched off.
> Anyway
> The settings of these three selectors is maintained between invocations,
> so if e.g. the if gain selection is on selection of the device not what it
> was
> when closing the program or switching to another device then that is
> definitely wrong
>
> The values are (should be) stored in the ".ini" file, if that file does
> not contain iitial values, then defaults will be set
> Just to be sure
>
> The selector for the agc is a checkbox left, on the 5-th line
> the if gain reduction selector is a spiinbox on the right hand side,
> allowing to set a range 20 .. 59,
> the lna state selector a spinbox, just below the if gain reduction
> selector. That one has for the band III 10 possible values, the
> effective reduction caused by the lna gain reduction is writtem to the
> right of this selector

In my code for the Uno I found a couple of things that might have been
wrong, esp. the handling of the antenna selector
There should not be an antenna selector since the only port for DAB (170 ..
230 MHz) is available for the tuner 1
I uploaded a 6.X version with a modified handler

andimik, 2024-05-31:
https://github.com/JvanKatwijk/qt-dab/blob/33daa11aa98026bdfb88f65f46537d9441475e76/i18n/de_DE.ts#L2199-2202

I guess it was my fault, he means `gain reduction`.

Frank-56, 2024-05-31:
You made it, Jan! With the new version RSPduo works as perfectly with qt-dab as the RSPdx. Thanks a lot! 

If you're located near Utrecht or Amstelveen I'll spend you a beer when I visit my before-retiring-colleagues there next time!

Cheers
Frank

JvanKatwijk, 2024-05-31:
Good to hear,

No I am not close to either Utrecht or Amstelveen. Long time ago I grew up
in Amstelveen (which obviously was much smaller than today).
Just for my curiosity, what are you retiring from?
(you obviously might answer on my private mail ***@***.***)

closed 2024-10-17 by JvanKatwijk