JvanKatwijk/qt-dab #330: Bug report Linux version 6.5

evertverduin, 2024-09-29:
Dear Jan,

I like your software, it works very nice. 
I found a small issue:
When selecting control:
Middle soundcard drop downbox has wrong index.

In my case i have to choose one soundcard further in the dropdown to get audio on the proper card ( -1 index ).
So it seems the index is 1 to much. 

BR

Evert

JvanKatwijk, 2024-09-30:
which sound system are you using? Qt_Audio or portaudio?

evertverduin, 2024-09-30:
> Hi Jan,
>
> It seems to be using Alsa.
>
> Here is the startuplogging:
>
> #BEGINLOGGING
>
> QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
> main: Detected system language "en_US"
> main: Set language "en_US"
> main: Error while loading language specifics "en_US" use English "en_GB" instead
> ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
> ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
> ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
> ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
> ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
> ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
> ALSA lib pcm_route.c:869:(find_matching_chmap) Found no matching channel map
> ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
> ALSA lib pcm_oss.c:377:(_snd_pcm_oss_open) Unknown field port
> ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection refused
>
> ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Connection refused
>
> [... 113 lines cut ...]
> OK, functions seem to be loaded
>
> #ENDLOGGING
>
> Verzonden met [Proton Mail](https://proton.me/mail/home) beveiligde e-mail.

BTW,

I am using the babyface as soundcard. But in order to select the RME babyface card, i have to select the lower Nvidia hdmi 0.

Br

Evert

Verzonden met [Proton Mail](https://proton.me/mail/home) beveiligde e-mail.

JvanKatwijk, 2024-10-01:
OK,
you are using the Linux version and the portaudio interface
Yes, tou are right, in the device list I start counting with one rather
than zero (long time ago the top element in the
combobox contained a label, something like "sound" or "sound device"

I'll create corrected versions later today

JvanKatwijk, 2024-10-01:
uploaded versions for 6.X with correct mapping in the audio device
selection for the portaudio audio system

evertverduin, 2024-10-05:
Hi Jan,

With the current RC release, the issue is still there. I pulled freshly and `compiled. The index shift is still there.

Br

Evert

JvanKatwijk, 2024-10-06:
Just be sure: the mod took place in the 6.X version, the 6.5 version is
more or less obsolete

JvanKatwijk, 2024-10-06:
https://www.dropbox.com/scl/fi/8rfim8f81gtydlff2nncb/Qt_DAB-6.X-x86_64.AppImage?rlkey=l5yjfx4izlxk6qkqtcx4rm856&st=dghz7jxn&dl=0

try this one

JvanKatwijk, 2024-10-06:
https://www.dropbox.com/scl/fi/8rfim8f81gtydlff2nncb/Qt_DAB-6.X-x86_64.AppImage?rlkey=l5yjfx4izlxk6qkqtcx4rm856&st=zh5k0mav&dl=0

evertverduin, 2024-10-06:
Hi Jan ( Hoi Jan ),

The AppImage works without any flaws.
Stuttering is also gone, index is correct.

I will be using the appimage. Is somewhat more convienient.

Best regards,

Evert

Verzonden met [Proton Mail](https://proton.me/mail/home) beveiligde e-mail.

closed 2024-10-06 by evertverduin