JvanKatwijk/qt-dab #332: Stuttering audio on linux 24-04-01 LTS

evertverduin, 2024-10-05:
Hi Jan,

I upgraded to 24-04-01 LTS and experienced stuttering and half bandwidth audio which is very specific to certain settings:

When stutter: ( mux 12C Zoetermeer ).
- If audio channel in mux is 72KBit/s 48000khz,  PS-SBR. Looks/sounds like a sampling issue.

When half bandwidth: ( mux 12C Zoetermeer ).
- If audio channel in mux is 72KBit/s, 32000khz, SBR only. Sounds like 8 KHz audio. Looks like sampling rate issue.

When normal: ( mux 12C national ).
- If audio channel in mux is 112KBit/s, 48000khz. SBR only. Sounds problemless

These problems started from ubuntui release 22-04-01. on 20-04-01 there were no issues.
THis is the compiled RC candidate, fresh pull from yesterday.

BR

Evert

JvanKatwijk, 2024-10-06:
Stuttering can have many causes, so I need more information
a. does the technical window shows any stuttering in the quality indicators
b. does it happen on all relevant audio channels
c. does it happen for both audio subsystems (i.e. portaudio and qtaudio)
d. does the stuttering happens immediately or after some time
e. is the stuttering continuous
f. what does control widget say about the cpu load
g. is the EPG handler running in the background

As a test I just ran the appimage on U24 running in a VM (only 4 cores) on
samples with 48, 72, 144 bits data
and listening for  a few minutes I do not hear stuttering

evertverduin, 2024-10-06:
After using the AppImage, the stuttering is gone. So, i suggest when someone runs into issues, use the appimage.

closed 2024-10-06 by evertverduin