JvanKatwijk/qt-dab #333: Strange sound while mono signal using FDK-AAC decoder

tomneda, 2024-10-11:
If found an issue here in that line:

https://github.com/JvanKatwijk/qt-dab/blob/ecfe40085516f454b21a71dff53b2ec97a751b68/src/backend/audio/fdk-aac.cpp#L107

The right channel got only each second sample and this sounds "interesting" :-)

So changing `bufp [2 * i]` to` bufp [i]` should work.

tomneda, 2024-10-12:
I forgot to mention the more critical thing: For me there is a memory overflow happen with reserving too little buffer memory here:
https://github.com/JvanKatwijk/qt-dab/blob/ecfe40085516f454b21a71dff53b2ec97a751b68/src/backend/audio/fdk-aac.cpp#L103

`alloca (2 * info -> frameSize)` reserves the memory on the stack in `bytes` but the data content is structured in `int16_t`. The number 2 already given is intended for the both stereo channels I guess.

So, for me the code should be adapted like this:
`int16_t *buffer = (int16_t *)alloca (2 * info -> frameSize * sizeof(int16_t));`

JvanKatwijk, 2024-10-14:
Thank you for the interest in Qt-DAB
Since in my environment there are no mono transmissions and I do not have
recordings with (DAB+) mono transmissions I cannot
reconstruct the phenomenon you mention.
It would be helpful if you could send a link to a recording with a (DAB+)
mono transmission so I can do my own investigation

JvanKatwijk, 2024-10-14:
As far as I can see in the code fragment you are referring to, but it
differs from the code in the repository

tomneda, 2024-10-14:
@JvanKatwijk : You are right, I linked somehow to an older version of that file. On the mainline head there is no buffer overflow issue anymore  with 
`int16_t buffer [2 * info -> frameSize];`

But your change to 
`buffer [2 * i + 1] = 0;`
would lead to only a sound on the left channel I, guess.

For a demo file I have to open a free cloud service, but it is currently too late for this. I can try it tomorrow. I can test your changes also for you, too.

tomneda, 2024-10-15:
I made a about 30s long uff file you find here for 3 days:

https://we.tl/t-5EJvSVWIL8

Look at service "BR24" for the mono signal.

With your current AppImage I hear the mono signal only the left channel as expected above.

JvanKatwijk, 2024-10-17:
The services BR24 and VR verkehr are mono and sound well - no problems -
whatsoever,
The sound quality of BR Verkehr is what you can expect with a bitrate of 16k
BR24 sounds well

JvanKatwijk, 2024-10-17:
https://www.dropbox.com/scl/fi/en9enh3d0psj6eo1gg11l/Qt_DAB-6.X-x86_64.AppImage?rlkey=1xgmtwkchklhriwdxq5z4rw3z&dl=0

As said, no problem with mini transmissions. Look for yourself with this
appImage

tomneda, 2024-10-20:
@JvanKatwijk, unfurtunately, the link above is not (more) working. I guess I waited too long for getting the file. Was not able to look into here the last days.

The main issue was the "interesting sound", means the twice as high sound speed on the right channel which you already fixed in your last changes: [Link](https://github.com/JvanKatwijk/qt-dab/blame/440070c4fb3791c387e6428b660e82d6c65864c8/src/backend/audio/fdk-aac.cpp#L110)

Setting the sample for the right channel to zero instead means that you only hear a sound in the left channel. But this may be your intention.

As the initial issue is solved you may close this issue.

closed 2024-10-21 by JvanKatwijk