JvanKatwijk/qt-dab #334: SpectrumScope window  Qt_DAB-6.X-x86_64.AppImage crashes

computer-wizard, 2024-10-15:
OS: Ubuntu 24.04.1 LTS
Qt_DAB-6.X-x86_64.AppImage version as released on October 11, 2024
Rx hardware: RTL2832U stick

Spectrumscope window  Qt_DAB-6.X-x86_64.AppImage crashes

This windows crashes very quickly, just when it syncs to the signal as it looks like, when I put my DAB signal on the air. I can 
restart that window but it crashes again. QT-DAB keeps working, only that spectrumscope window quits with a signal.  I see nothing in the log or on stderr about this.

The spectrumscope window stays up when I up put the receiver on another frequency without signal.

JvanKatwijk, 2024-10-15:
> You are absolutely right, "interesting" phenomenon. Since the software doe
> not tell what happens, I'll instrument the spectrum widget
> to get some information, but as far as I recall, after compiling the
> software on U24 there was no probllem. I'll have some homework to do

Just a quick check: a freshly built version on a U24 VM with the Qt5
libraries showed the same behaviour, although the spectrum showed for
a longer time.
Anyway, I 'll have to find out what is happening on U24

JvanKatwijk, 2024-10-15:
It was an interesting kind of phenomenon. Not sure it was an error, bit the
vause is removed. I uploaded greshly created precompiled versions with this
problem removed.

closed 2024-10-17 by JvanKatwijk