JvanKatwijk/qt-dab #335: Problem with latest Win64 installer

davidbstanley, 2024-10-15:
The Windows 64 bit installer is failing for the latest version 6.5.  The error says 'C:\Program Files (x86)\qt-dab-64\qt-dab64-6.X.exe: An error occurred while trying to copy a file: The source file is corrupted'.

The beta version has installed without this issue.

JvanKatwijk, 2024-10-15:
I did not have problems with installing, but since I had to remove a
potential error (not related to installing) I created new precompiled
versions.
The 64 bit version was tested on two windows boxes, so if there is still a
prob lem it might have yo do with something going wrong while downloading

davidbstanley, 2024-10-15:
I have just downloaded again and whatever you did solved the problem.  All fine now.