JvanKatwijk/qt-dab #340: qt-dab 6.9 beta - audio breaks up after a few minutes

Philip-radioamateur, 2024-12-22:
Hello Jan,

Thank you very much for continuing to develop your excellent qt-dab.

I hope it will useful to you if I let you know about two issues I am experiencing with your latest release of 21/12/24:

1.  After a few minutes with good audio the audio breaks up into bursts with gaps ('stutters').  By changing to another multiplex and then back to the original multiplex the problem is fixed for a few minutes but then the audio stutters again.  This happens on all multiplexes and was not a problem with your earlier 6.9 beta release.

2.  The 'dx display' window vertical scroll bar does not work.

My set up is an RSP1A and Windows 10 desktop PC.

I am in London.

Thanks again for all of your good work on your software.

Philip.

JvanKatwijk, 2024-12-22:
Thanks for the info, I'll have a look

JvanKatwijk, 2024-12-23:
I gad Qt-DAB run on a w11 box for about an hour without problem with the
sound.
It might be the case that the new TII detector is too demanding when the
threshold  is too low.

Wrt the dxdDisplay: that seems to be a Qt problem, I have to look into it

best
jan

Philip-radioamateur, 2024-12-23:
Thanks for your reply Jan,

After reading your reply, I tried setting TII to Local and the audio is OK.  I then went back to DX Display and gradually increased the threshold from 4 dB and found that 8 db gives clear audio.

Philip.

JvanKatwijk, 2024-12-24:
I see that something is wrong.
You have selected the portaudio audiooutput, I did not have that and could
not see  the problem, after selecting the portaudio output I see
that after a few minutes the sound drops, and it returns a minute later.
I'll have to look into that

best
jan