JvanKatwijk/qt-dab #345: Webserver not displaying transmitters

brokenbadger, 2025-01-16:
Hi,

Been scratching my head trying to figure this one out???
I've downloaded the seperate dbloader program and run it to get the output file.
Moved the file to the qt-DAB-files folder (made sure the path was specified).
Entered my coordinates and made sure http-on was selected.
Opened up a browser and went to localhost:8080 (port was already assigned to 8080).

All I get is a map with my location and no lists of transmitters or symbols? Doesnt matter which control option I try?
It seems as though the output file is just not being read?

Am I missing something here? I couldnt find any direct instructions on setting this bit up?
Apologies in advance if I'm being a noob.

Awesome project by the way, thank you for creating this.

JvanKatwijk, 2025-01-16:
Do not move the downloaded db, it belongs in your home folder where it is
placed after downloading

brokenbadger, 2025-01-16:
Yup, thats fixed it, thanks.
I did try with it there originally with no results (which is why I thought I needed to move it?), but with hind sight it may have been my browser blocking an element? Which I have now addressed.
That moment when you know it's something simple, but you can't see it for love nore money? 

Thanks for your patience, appreceated.

closed 2025-02-06 by brokenbadger