JvanKatwijk/qt-dab #346: Ricerca canali Dab

dab983, 2025-01-18:
In the evening I installed qt6-dab64-6.9 and also qt-db-loader on the folder automatically chosen by the program, on my PC with Windows 10, it recognizes the RTL SDR COM stick, I used Zadic, but it doesn't find any DAB channels, can anyone help me? step tell me the various operations to do? Thank you

dab983, 2025-01-18:
In the evening I installed qt6-dab64-6.9 and also qt-db-loader on the folder automatically chosen by the program, on my PC with Windows 10, it recognizes the RTL SDR COM stick, I used Zadic, but it doesn't find any DAB channels, can anyone help me? step tell me the various operations to do? Thank you

JvanKatwijk, 2025-01-19:
a. I assume you have an antenna
b. the stick is recognized by the software, what are your gain settings
c. what channels do you know have DAB content
d. what do you see as spectrum

Just send a picture than we might get an idea

andimik, 2025-03-13:
@dab983 

Could you please answer Jan's questions? I assume you are from Italy, so probably you try the channel from DAB Italia, which has the best coverage in the country.

If you open the spectrum window, you might help yourself then.