JvanKatwijk/qt-dab #349: Compiling problem on 6.9 with Raspberry Pi 4 (Bookworm)

ocean71, 2025-02-01:
I hope you can help me.

I am having problems compiling the latest 6.9 for my Pi4. I installed 6.8 OK.
I downloaded and installed qwt 6.3.0 as per the instructions and installed the required libraries including the Qt6 ones. The FindQwt module file was amended accordingly to reflect correct paths. 

Using Cmake, I manage to get 74% of the compiling when it throws up an error. 

I would be very grateful for some advice on how to sort the problem out.

This is what happens:

[ 74%] Building CXX object CMakeFiles/Qt-DAB.dir/home/xxxxxx/Downloads/qt-dab-Qt-DAB-6.9/src/output/Qt6/Qt-audio.cpp.o
/home/xxxxxx/Downloads/qt-dab-Qt-DAB-6.9/src/output/Qt6/Qt-audio.cpp: In member function ‘virtual void Qt_Audio::restart()’:
/home/xxxxxx/Downloads/qt-dab-Qt-DAB-6.9/src/output/Qt6/Qt-audio.cpp:110:9: error: ‘QtAudio’ has not been declared
  110 |         QtAudio::Error err = m_audioSink -> error ();
      |         ^~~~~~~
/home/xxxxxx/Downloads/qt-dab-Qt-DAB-6.9/src/output/Qt6/Qt-audio.cpp:111:50: error: ‘err’ was not declared in this scope; did you mean ‘erf’?
  111 |         fprintf (stderr, "Errorcode %d\n", (int)(err));
      |                                                  ^~~
      |                                                  erf
make[2]: *** [CMakeFiles/Qt-DAB.dir/build.make:2095: CMakeFiles/Qt-DAB.dir/home/xxxxxx/Downloads/qt-dab-Qt-DAB-6.9/src/output/Qt6/Qt-audio.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:102: CMakeFiles/Qt-DAB.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

I hope someone can solve this for me. Many thanks for your help.

JvanKatwijk, 2025-02-03:
I am a little confused about the error,

In your message the erroneous lines seem to point to line 110 and 111 of
the file Qt6/Qt-audio.cpp.
These lines don't refer to either the symbols QtAudio or err,

It seems the code is referring to the "restart" function, which is located
on the lines 83 .. 103.
In this function there is an occurrence of a symbol err, and symbol
QtAudio, the latter referring to a class in Qt6.
If that class cannot be found, you most likely get the errors you had
(QtAudio class contains an Error variable which on line 101 is used to
declare
a variable err

ocean71, 2025-02-03:
Thank you very much for your reply.

As I think I have followed the installation instructions and installed all the libraries, it looks as though I won't be able to install 6.9 on the Pi. I do not know much about coding, so I would not be able to track down the source of the problem myself.  

As an aside I did try qmake as well but that failed very early on in the compiling process.  With this message:
 WARNING: Failure to find: ../includes/support/settingnames.h

Not to worry - I will continue to use 6.8 which installed with no problems using Cmake.
Thank you for looking into this.
Best wishes.

JvanKatwijk, 2025-02-03:
IKe, the settingnames is a typo, the file reads settingNames.h. Interesting
both Ubuntu and fedora find the file

ocean71, 2025-02-04:
Thank you very much for letting me know about the settingNames issue.
I made that alteration and ran qmake again. But without success.

Below I list the result. I have also included a section of the .pro file which I have amended in an attempt to allow for a Pi4 running an RTLSDR dongle, with qwt 6.3.0 also installed as per the instructions.

My guess is that I have not made the correct changes to this .pro file. But I also wonder if I have the correct Qt-audio.cpp file as you refer to a comment in line 101 in your version, but in the one I am using the line is blank. This was part of the Source Code which I downloaded  from the Release Section of 6.9.

Anyway, below is a record of what happened:

**Using ‘qmake ..’ and then ‘make’ on 6.9 using Source Code provided on the Release Page**

.pro file altered as follows for Pi 4 using RTLSDR dongle:   

Lines 433-476

INCLUDEPATH	+= /usr/include
!mac {
INCLUDEPATH	+= /usr/include/qwt-6.3.0/lib
#correct this for the correct path to the qwt6 library on your system
#LIBS		+= -lqwt
equals (QT_MAJOR_VERSION, 6) {
	 LIBS		+= -lqwt-qt6
	}else{  LIBS += -lqwt-qt5
	}
}

#mac {
# Should be possible to make on non Macs as well.
# qmake -set QMAKEFEATURES /usr/local/Cellar/qwt/6.2.0/features
CONFIG		+= qwt
#}
#CONFIG		+= double
CONFIG		+= single
#
# comment or uncomment for the devices you want to have support for
# (you obviously have libraries installed for the selected ones)
#CONFIG		+= sdrplay-v2
#CONFIG		+= sdrplay-v3
CONFIG		+= dabstick-linux
#CONFIG		+= rtl_tcp
#CONFIG		+= airspy-2
#CONFIG		+= hackrf
#CONFIG		+= lime
#CONFIG		+= soapy
#CONFIG		+= pluto-rxtx
#CONFIG		+= pluto
#CONFIG		+= spyServer-16
#CONFIG		+= spyServer-8
#CONFIG		+= uhd
#CONFIG		+= colibri
#CONFIG		+= elad-device
CONFIG		+= faad
#CONFIG		+= fdk-aac
#CONFIG		+= preCompiled
CONFIG		+= tiiLib
#very experimental, simple server for connecting to a tdc handler
CONFIG		+= datastreamer
#to handle output of embedded an IP data stream, uncomment
#CONFIG		+= send_datagram

RESULT of running Qmake (after creating 303 files in the build folder)

/usr/bin/ld: spiral-no-sse.o: in function `FULL_SPIRAL_no_sse':
./../../src/support/viterbi-spiral/spiral-no-sse.c:26: multiple definition of `FULL_SPIRAL_no_sse'; spiral-no-sse.o:./../../src/support/viterbi-spiral/spiral-no-sse.c:26: first defined here
collect2: error: ld returned 1 exit status
make: *** [Makefile:874: linux-bin/qt-dab-qt5-6.9] Error 1

Makefile line 874 is as follows

	$(LINK) $(LFLAGS) -o $(TARGET) $(OBJECTS) $(OBJCOMP) $(LIBS)

Because of this failure to compile and the possible issue with the Qt-audio file mentioned above , I also downloaded the 6.9 master from the Code page (rather than the Release section) but that also failed on Qmake.  

Using 6.9 Master
tCore -I. -I. -I/usr/lib/aarch64-linux-gnu/qt5/mkspecs/linux-g++ -o main.o ../main.cpp
g++: error: unrecognized command-line option ‘-mavx2’
g++: error: unrecognized command-line option ‘-msse4’
make: *** [Makefile:1928: main.o] Error 1

Sorry for such a long post but I wanted to provide as much information as I could.

Please don't worry if you do not have the time to study this - I can always use 6.8, so please close this if that is the case. I will understand.
Best wishes.

ocean71, 2025-02-04:
Sorry about the strange formatting above - don't know why that happened!

JvanKatwijk, 2025-02-04:
#CONFIG         += faad
CONFIG          += fdk-aac
#very experimental, simple server for connecting to a tdc handler
CONFIG          += datastreamer
#to handle output of embedded an IP data stream, uncomment
#CONFIG         += send_datagram

#if you want to listen remote, uncomment
#CONFIG         += tcp-streamer         # use for remote listening
#otherwise, if you want to use the default, uncomment
CONFIG          += local-audio

CONFIG          += viterbi-new
#CONFIG         += spiral-sse
#CONFIG         += spiral-no-sse
#DEFINES        += SHOW_MISSING
DEFINES         += __LOGGING__
DEFINES         += __DUMP_SNR__         # for experiments only

You should uncomment "viterbi-new" and comment out "siral-xxx_ versions

In the section viterbi-new commetn out everything that has to do with X86
boxes and even the AARCH64, since that may depend on
other include files.
The basic version for viterbi is a scalar version, which takes more time to
execute but works fine

iterbi-new {
        DEPENDPATH      += ../src/support/viterbi
       #QMAKE_CFLAGS    +=  -mavx2 -msse4
        #QMAKE_CXXFLAGS  +=  -mavx2 -msse4
        #QMAKE_LFLAGS    +=  -mavx2
        #DEFINES         += __ARCH_X86__
        #DEFINES        += __SSE4_1__
        #DEFINES        += __AVX2__
        #DEFINES        += __ARCH_AARCH64__
        INCLUDEPATH     += ../src/support/viterbi
        HEADERS         += ../src/support/viterbi/viterbi.h
        SOURCES         += ../src/support/viterbi/viterbi.cpp
}

The message you got was related to the spiral implementation of the viterbi
decoder, but I suggest the use of the "new"
version

ocean71, 2025-02-04:
Thanks very much for the quick response. I incorporated the changes you suggested and ran qmake again.

Unfortunately I got this error. Not sure what to do with this one?

../../src/support/viterbi/viterbi.cpp:39:10: fatal error: emmintrin.h: No such file or directory
   39 | #include <emmintrin.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:3868: viterbi.o] Error 1

Thanks for your help

JvanKatwijk, 2025-02-05:
In your case you can comment the lines
//#include <emmintrin.h>
//#include <xmmintrin.h>
//#include <mmintrin.h>

just out, they relate to sse type support

ocean71, 2025-02-05:
Thanks for your latest note and help.
Yes, it now compiles fine on the Pi with the latest amendments.  I am very grateful for your help and patience - and your perseverance.

Thanks again, and thank you of course for producing such a useful program.
Best wishes.

closed 2025-02-05 by ocean71

JvanKatwijk, 2025-02-05:
can you send me some details on the problems you encountered, might be
helpful for others as well

best
jan

ocean71, 2025-02-05:
Of course. Happy to do so. Should I post here or send it to you by email?

JvanKatwijk, 2025-02-05:
either way

ocean71, 2025-02-06:
Sorry to trouble you again. Having got everything working, I am not sure how to add the tii database. The appimage does not work on my pi.

Maybe this database is no good on a pi?

Thanks

JvanKatwijk, 2025-02-07:
> Once I had downloaded the qt-dab-master from the code page,
> installed all the libraries listed in the Readme on the code page,
> and then installed the new qwt 6.30 as instructed,
> I had a problem getting the .pro file to recognise the new qwt version.
> This was because on the pi it installs to /usr/include.
>
> So I changed the .pro file as follows:
> Line 433
> INCLUDEPATH += /usr/include
> !mac {
> INCLUDEPATH += /usr/include/qwt-6.3.0/lib
> #correct this for the correct path to the qwt6 library on your system
> #LIBS += -lqwt
> equals (QT_MAJOR_VERSION, 6) {
> LIBS += -lqwt-qt6
> }else{ LIBS += -lqwt-qt5
>
> However I still could not get the qwt recognised so further changes were
> needed, this time to the Modules folder.
>
> To do this, I went to the qt-dab-6.9 folder in the qt-dab-master, and then
> opened the folder cmake. Then Modules.
> Found the relevant FindQwt file and made the following changes:
>
> find_path(QWT_INCLUDE_DIRS
> [... 39 lines cut ...]
> So I made the recommended changes mentioned earlier in this thread one by
> one and it compiled with no problems.
>
> Hope this helps to point people in the right direction and that I
> remembered everything!

Thanks, I' will add the comment to the website

JvanKatwijk, 2025-02-08:
Sorry, I missed this one.
Download the TII database with this link
https://www.dropbox.com/scl/fi/x2v8vv3728a3vm3wactuo/txdata.tii?rlkey=ogor7yrsrx8mrmy6b8ztoorsy&dl=0

After downloading rename the file ".txdata.tii" and place it in the user's
home directory.
Qt-DAB will look for the database there

JvanKatwijk, 2025-02-08:
https://www.dropbox.com/scl/fi/x2v8vv3728a3vm3wactuo/txdata.tii?rlkey=ogor7yrsrx8mrmy6b8ztoorsy&dl=0

I probably uined the previous link

ocean71, 2025-02-08:
Thanks very much indeed for the file. Works fine.
Thanks a lot for your help with this.
Best wishes.