JvanKatwijk/qt-dab #350: Checkout issue

gvanem, 2025-02-07:
A problem with a `git clone` of this project.
I'm on Win-10 and did:
```bat
md Git-latest
cd Git-latest
git clone https://github.com/JvanKatwijk/qt-dab.git .
Cloning into '.'...
remote: Enumerating objects: 20424, done.
remote: Counting objects: 100% (1975/1975), done.
remote: Compressing objects: 100% (689/689), done.
remote: Total 20424 (delta 1471), reused 1393 (delta 1286), pack-reused 18449 (from 3)
Receiving objects: 100% (20424/20424), 549.85 MiB | 29.33 MiB/s, done.
Resolving deltas: 100% (15614/15614), done.
error: invalid path 'includes/backend/*.cpp'
fatal: unable to checkout working tree
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/'
```
And a `git restore --source=HEAD :/` also fails.

What's this? A path/filename issue on Windows only?

JvanKatwijk, 2025-02-07:
I never work with Windows (other than crosscompiling and testing).
cloning the sources works fine on Linux

gvanem, 2025-02-08:
> I never work with Windows (other than crosscompiling and testing).

Yet you want Qt-DAB to work on Windows? 
When downloading a .zip from this repo, WinZip reports this:
[image]

And `7z l qt-dab-master.zip` shows:
```
...
2025-02-05 12:52:51 D....            0            0  qt-dab-master\includes\backend
2025-02-05 12:52:51 .....            0            0  qt-dab-master\includes\backend\*.cpp
...
```
There should be no `*.cpp` files under `includes/`, no? You have done a bad git command AFAICS.  

Please fix this.

JvanKatwijk, 2025-02-08:
Thank you for your friendly comment, please recall that if you are not
satisfied with the software "as is",
ask your money back and stop using the software

closed 2025-06-12 by JvanKatwijk