JvanKatwijk/qt-dab #352: qt-dab does not load a new .txdata.tii

Frank-56, 2025-02-14:
Hi Jan,

I'm not quite sure about which version I'm talking, it installs in qt-dab64-6.**8**, but shows qt-dab64-6.**9** in the main widget, so I suppose 6.9 ...

"refresh table" in 6.9/64 and 6,9/32 does nothing, while "load table" in 6.x/32 and 6.8/32 works well and a messagebox tells me a new table has been loaded. No new .txdata.tii in 6.9, new .txdata.tii in 6.8.

Cheers
Frank

JvanKatwijk, 2025-02-16:
T\The "refresh table" button assumed a database is in the user's
homedirectory - loaded by the db-loader which is now a separate program - a
then refreshes the copy of the database the system is running with

Frank-56, 2025-02-16:
Thanks Jan,

was not clear to me. Now works of course.

But there is a general issue with getting new data from fmlist: In case a VPN like Cyberghost or so is active, fmlist denies to give new data and the resulting ,txdata.tii is only 100 bytes large, although it says "Download succeeded". Qirx faces the same issue and fmlist denies me to log in when running with a VPN.

When I click "Push voor download" more than one time, the .txdata.tii gets larger and larger on each click, think you append. Only when I close dbloader and start it again, it has the initial size of 705k at the moment.

There more on tii to come, but I must fully figure out first.

JvanKatwijk, 2025-02-17:
The very first version of the dbloader had that probem, in later versions
that was solved (I think, I;; have a look)

JvanKatwijk, 2025-02-17:
https://www.dropbox.com/scl/fi/9lgn26petyl5a40vzdvri/setup-qt-db-loader.exe?rlkey=bgz1i0wevmnwj1lic67xuurvv&dl=0

probeer deze is, die geeft hier een resultaat van iets meer dan 77000,
ongeacht het aantal keren dat je hem aanklikt

closed 2025-06-12 by JvanKatwijk