JvanKatwijk/qt-dab #355: fatal error: xml-handler.h: No such file or directory - Missing dependency?

hvegh, 2025-03-04:
While buiding in  qt-dab-6.9/build :

cmake ..
...
[ 65%] Building CXX object CMakeFiles/Qt-DAB.dir/home/henk/prj/sdr/qt-dab/src/support/findfilenames.cpp.o
[ 66%] Building CXX object CMakeFiles/Qt-DAB.dir/home/henk/prj/sdr/qt-dab/src/support/content-table.cpp.o
[ 66%] Building CXX object CMakeFiles/Qt-DAB.dir/home/henk/prj/sdr/qt-dab/src/support/ITU_Region_1.cpp.o
/home/henk/prj/sdr/qt-dab/src/support/time-table.cpp:27:17: fatal error: xml-handler.h: No such file or directory
   27 | #include        "xml-handler.h"
      |                 ^~~~~~~~~~~~~~~

File is in:

../../qt-devices/sdrplay-handler-v2/xml-handler.h

Any pointers what is wrong?

JvanKatwijk, 2025-03-04:
You are using CMake. I always use qmake. It was as such an interesting
error, showing the differing in how cmake
and qmake operate.
The files xml-handler.... in the sdrplay_2 environment have norhing to do
with it and therir function was replaced quite some time
ago by the xml-filewriter
The reference to xml-handler in the file ../src/support/time-table.cpp is
the culprit, just remove the line and there should not be
a problem with compilation.
One remark though, there is - for many years now - no support from SDRplay
for the 2.13 library, and there is actually no reason that
tje 2.13 library is still in the cmake and the qmake route available. I
strongly suggest using the 3.1X route

best
jan

hvegh, 2025-03-06:
qmake6 .. did the trick!
As was mentioned in the documentation, i should have read more carefully....

also I had to patch:
```patch
diff --git a/qt-dab-6.9/qt-dab-6.9.pro b/qt-dab-6.9/qt-dab-6.9.pro
index 2a008ab9..2a300858 100644
--- a/qt-dab-6.9/qt-dab-6.9.pro
+++ b/qt-dab-6.9/qt-dab-6.9.pro
@@ -434,7 +434,7 @@ INCLUDEPATH += /usr/local/include
 #correct this for the correct path to the qwt6 library on your system
 #LIBS          += -lqwt
 equals (QT_MAJOR_VERSION, 6) {
-        LIBS           += -lqwt-qt6
+        LIBS           += -lqwt6-qt6
        }else{  LIBS += -lqwt-qt5
        }
 }

```

Thank you both  for the comments!

closed 2025-03-06 by hvegh