JvanKatwijk/qt-dab #357: qt dab won't run ?

m1ctk, 2025-03-13:
as title

andimik, 2025-03-13:
I am pretty sure, Jan will close this issue without any comment. Next time, please find the time to write more than this useless and annoying headline.

And read the manual.

The software works well under Linux and Windows on several laptops and PCs in my home.

m1ctk, 2025-03-14:
Don't be rude mate. I've added the version no, earlier versions like 6.6 use to run no problem but this one won't run for some reason using Windows 10 64 bit

JvanKatwijk, 2025-03-14:
Can you be slightly more specific
does it mean
a. the program does not start at all
b. the program starts, but after showing the widgets, it spontaneously halts
c. the program starts and selecting a channel is possible but no spectrum
is seen
d.  the progra starts, a channel is selected and a list of services shows
but selecting a service does not have effect
etc etc

m1ctk, 2025-03-21:
For some reason 6.9.2 runs fine though I am afraid it is still deaf compared to the excellent AbracaDABra and QIRX

JvanKatwijk, 2025-03-21:
are you using an older dabstick? this version of qt dab was built with the
V4 library

m1ctk, 2025-03-23:
> are you using an older dabstick? this version of qt dab was built with the
> V4 library
> 
> 
> Op vr 21 mrt 2025 om 20:17 schreef m1ctk ***@***.***>:
> […](#)

I use an rspDX

closed 2025-06-12 by JvanKatwijk