JvanKatwijk/qt-dab #361: Save MP2

Confused-52, 2025-07-27:
Trying 6.9.2 - the previous versions I tested stored mp2 when tuned to Dab and aac when tuned to Dab+ when the save AAC button was pressed. That seems to me to be the correct behaviour although the button text could be better name as save coded audio. Is there any chance of getting that feature back? The installed base of Dab radios in the UK owned by elderly people is probably such that some stations on National multiplexes will remain as classic DAB to avoid losing their audience.

JvanKatwijk, 2025-07-29:
I do not recal;l changing it for quite some time. You want the MP2 output
(the "segments" stored into a file? Dumping the MP2 segments into a file is
not very compilated, I'll have a look
best
jan

JvanKatwijk, 2025-07-30:
You could try the uploaded 6.9.3 version (I assume you need the windows version). As far as I can tell it works fine, saving MP2 files. One (small) thing, the file extension is ".mp2" but my windows box calls the MP2 files "MP3".
They can be played with a player though.
I appreciate if you report back whether it works or not

Confused-52, 2025-08-02:
New stick arrived this morning and the temp fix version works on the RTL-SDR v4  thank you. It also works on a schedule and receives the xml EPG ok too. Thanks very much, I did not expect such a fast and comprehensive response. Thanks again.