JvanKatwijk/qt-dab #362: Accessibility feedback for Qt-DAB with NVDA screen reader

gsasa, 2025-09-09:
Hello,
First of all, thank you for developing Qt-DAB — it is a great lightweight DAB+ receiver application and works very well with RTL-SDR devices.
I am a blind user and access the program with the NVDA screen reader. Qt-DAB is already almost fully accessible, but I noticed one important issue:
• 
In the station list, I can move with the arrow keys, but pressing Enter or Space does not activate (play) the selected station.
• 
Once focus is inside the station list, it is also difficult to return to the playback controls by keyboard only.
If Enter or Space could be bound to the “activate/play selected service” action, and if Tab or F6 could move focus between the station list and the control area, Qt-DAB would become 100% accessible.
I hope you will consider adding these small improvements. They would make a big difference for blind and visually impaired radio enthusiasts.
Thank you very much for your work!
Best regards, Sasa.

JvanKatwijk, 2025-09-13:
You can give this version (testversion) a try
https://www.dropbox.com/scl/fi/t3thv40392xdg6o32j03t/setup-qt-dab32-qt6-6.9V3.4.exe?rlkey=38qpcj11xkami24y10bo4svrx&dl=0

best
jan

gsasa, 2025-09-13:
Thank you very much. On Monday I will be at the location where my SDR is and where I have the possibility to receive DAB+ signals. I will test this version and let you know.

gsasa, 2025-09-15:
Hello. Activating the Station with Taster Enter works great, but still cannot be moved by the application windows.

JvanKatwijk, 2025-09-15:
I am not sure I understand the issue. As far as I can see, the F1 key sets
the focus on the main window, the F2 key on the configuration window and
the F3 key on the technical data window
Of course if the focus is not on the main window, the up/down arrow do not
work on the services list
Is that what you mean?

gsasa, 2025-09-18:
Hello. The classic command for switching focus from widget to widget is f6. I suggest you go to nvaccess.org and download the free screen reader NVDA. It can also be portable. You do not need to install it. Open Google Chrome, Microsoft Edge, or This PC and press f6. You will see what happens.

gsasa, 2025-09-19:
If the widgets are a neatly named screen reader will neatly announce which widget or window is active and then the order is not important. Best regards,

gsasa, 2025-10-02:
Hi @JvanKatwijk I had a lot of some other obligations so I didn't arrive to download and test the program. Today I took over it and I'll test it, so I answer you the results.